Xinyuan-LilyGO/T-Display-S3-Pro #4: Touch not working reliable

pixelwave, 2023-12-31:
I have the issue that the touch screen is not working reliable on my T-Display-S3-Pro. The problem is with the factory code as well as my custom code using Arduino GFX + LVGL. When using the "capacity touch" sample code it seems to detect all inputs so it does not really seem like the device itself is broken? I also have a second unit - same behaviour?

https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/assets/19491804/4728e317-8e50-4639-a78a-a0311a4c3041

https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/assets/19491804/d52b48f9-dbf8-43fa-9f5b-b6b743c385fd

lewisxhe, 2023-12-31:
First confirm whether the touch is normal, you can write https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/examples/CapacitiveTouch to test.

pixelwave, 2023-12-31:
> First confirm whether the touch is normal, you can write https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/examples/CapacitiveTouch to test.

Did you read my post? That is what I already confirmed „.. when using the "capacity touch" sample code it seems to detect all inputs so it does not really seem like the device itself is broken“.

pixelwave, 2023-12-31:
Still troubleshooting … the issue seems to get worse when using battery instead of usb power?

lewisxhe, 2023-12-31:
When you apply it in your project, have you printed the touch output coordinates and are they normal? Or do you display it on the screen and observe whether the coordinates are the same as the actual touch position?

pixelwave, 2024-01-01:
So I can confirm it has something to do with battery vs usb powered mode.

You can try my sample code yourself (https://github.com/pixelwave/t-display-s3-pro) with the following test setups:

1) Power Switch "OFF", connected and powered from USB port (PC/Laptop)
2) Power Switch "ON" battery powered only

In scenario 1) the touch inputs work reliably and 99% of the time. In Scenario 2) the touch is a hit and miss. Sometimes the buttons work sometimes not. 

So in summary I assume a general hardware issue with the board design. I hope this can be resolved in an updated version (I have V1.1) because otherwise the product is quite awesome! But with that flaw currently unusable im battery mode. 

Contacting support to find a solution ..

ShallowGreen123, 2024-01-02:
Please tell me how many S3-Pro you bought in total, you can contact the after-sales service to return or replace the goods.

pixelwave, 2024-01-02:
I bought 2 in total already contacted sales (sales@lilygo.cc) and they forwarded it to the orders (orders@lilygo.cc) team. Currently waiting for their feedback. I noticed you are using a different unit? Is it maybe a preproduction or newer unit? Has there been any updates since V1.1? Because both my units show the same behaviour I tend to believe it is more a general hardware design issue? 

See below my production (V1.1) device with and without cable:

https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/assets/19491804/058bc7e9-1d75-4273-9e98-23b69ad9c587

https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/assets/19491804/4569dcfc-1853-47ad-8aff-7bc7a3949baa

lewisxhe, 2024-01-02:
I know what your problem is. Please change it to the following:
`
build_flags=
        ; -DARDUINO_USB_CDC_ON_BOOT=1
        -UARDUINO_USB_CDC_ON_BOOT
`

If Cdc is turned on, the serial port will block the program from running.

pixelwave, 2024-01-02:
The arguments are as written in the official lilygo sample code. 

But I compiled it with the variables like you suggested:
[image]

No change same behaviour. When in "battery mode" touch screen does NOT work reliably.

lewisxhe, 2024-01-03:
Please note that 
`
; -DARDUINO_USB_CDC_ON_BOOT=1 
`
has been commented out and
`
 -UARDUINO_USB_CDC_ON_BOOT 
`
is enabled

pixelwave, 2024-01-03:
Compiled now like this:

[image]

Still unchanged behaviour. USB powered = all fine. Battery powered = touch very janky.

lewisxhe, 2024-01-03:
If you can try to enter the binary file I provided, and test whether it is caused by environmental reasons.

The writing method can follow  [here](https://github.com/Xinyuan-LilyGO/LilyGo-Document/blob/master/MacFlashBin.md)

[t-display-s3-pro_0103.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/files/13816595/t-display-s3-pro_0103.zip)

pixelwave, 2024-01-03:
Flashed once with esptool and once with thonny. Same faulty behaviour.

pixelwave, 2024-01-03:
Flashing worked both times. I just wanted to try with different tools. Also both times completely erased flash.

The touch problem is still there.

lewisxhe, 2024-01-03:
I have received a message from customer service. I am very curious about what the problem is. Customer service will communicate with you next. I hope to send back these two boards. I want to find out the reason.

pixelwave, 2024-01-03:
Yes me too! Would love to ultimately fix the problem and have working boards. Because the product and paper is great. So I am going to contact with customer support next to send the boards back? I will leave the code on there for you to check ...

pixelwave, 2024-01-08:
... support is not answering anymore? I asked how to return the product. Can you help or contact them?

lewisxhe, 2024-01-08:
Haven't you already contacted customer service?

lewisxhe, 2024-01-08:
Tomorrow I will ask how to deal with it, and I will reply to you here after I get the answer.

pixelwave, 2024-01-08:
Thanks a lot!

lewisxhe, 2024-01-10:
> Thanks a lot!

Did customer service contact you?

pixelwave, 2024-01-10:
Did yesterday - thanks for pinging. But still waiting for their solution on how to ship back the products.

samarkh, 2024-02-25:
My unit also has the same issue, fine on USB power, but bad when self-powered.

pixelwave, 2024-02-25:
After all testing I really think it is a hardware based issue. Maybe just for this specific revision (mine was V1.1). @lewisxhe BTW I am still waiting for the shipping cost refund until I can return the two units for you to test ...

lewisxhe, 2024-03-05:
@pixelwave Hi, have you not arranged how to ship it back after the sale? It's a little weird that both of them have problems. If I can ship them back, I will check what caused the problem.

czm8, 2024-03-16:
I'm wondering if anyone knows if there is a solution for this. I recently came across the same issue - When I have the T-Display S3 Pro connected to my computer with a USB cable the touch screen reacts quickly and perfectly. However, when I disconnect the USB cable from the T-Display S3 Pro the touch because much less reliable and often doesn't sense my finger touching the screen. I have it set up so that it displays the x and y coordinates on the screen so I can see if it sees where I touched the screen and there I can see that the numbers do not change.

I have also recreated the problem I was seeing before - this time using the "CapacitiveTouch" example provided by LilyGo on their GITHUB page meaning I can rule out anyting in my code as being a problem. [https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/examples/CapacitiveTouch](https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/examples/CapacitiveTouch)

To summarize, the problem is a follows, when the unit is powered from the USB port, it recognizes all touches on the screen without any problems. When the USB power is removed and the unit is powered by the internal battery it sometimes works and sometimes doesn't recognize the touches on the screen. This makes the T Display S3 Pro unusable for me as a battery powered, portable device.

I have sent LilyGo an email about this issue but I still have not heard back from them. Oh and my unit is version 1.1

pixelwave, 2024-03-17:
> @pixelwave Hi, have you not arranged how to ship it back after the sale? It's a little weird that both of them have problems. If I can ship them back, I will check what caused the problem.

I wrote customer service now one last time. The faulty products are packed and ready to ship batteries removed and all as advised. But they still did not refund the return shipping cost. In fact it took so long DHL now raised fees for shipping. -.-

If you guys are still interested in receiving the faulty products I told them they have until March 22nd to respond - otherwise I will destroy them and close this case.

**But seeing now more and more cases popping up - again I believe it is probably a hardware issue of the entire batch / revision. So you should be able to also recreate this error if you have a hardware version V1.1 laying around ...**

pixelwave, 2024-04-09:
@lewisxhe 

Did the boards arrive by now? I saw some people have the same issue with the T-Display-S3-Pro. Could you maybe verify this with a hardware revision (V1.1) if you have units available locally?

lewisxhe, 2024-04-09:
I haven't received the package yet, it should still be on its way. I will test it as soon as I receive it.
All factory tests are conducted using batteries for power supply, and no such situation has been found so far

pixelwave, 2024-04-29:
Hi @lewisxhe, any updates so far?

lewisxhe, 2024-04-29:
Just inquired and it has arrived, I will test it as soon as I get it

lewisxhe, 2024-04-30:
@pixelwave I have tested the two boards you sent back. One of them is labeled with a defective device. I tested it and the touch was very insensitive when powered by battery. After testing, it became normal after replacing the screen, so I suspect it is. The sensitivity of the capacitive touch screen decreases when powered by battery, but works normally when powered by USB. I have reported the problem to the supplier.

Judging from the current tests, it is the touch sensitivity of the screen that causes the touch to be insensitive. I have added a touch screen test to the factory test steps. This should be avoided in the future. Thank you again for your feedback.

lewisxhe, 2024-04-30:
oh! By the way, another device without a label is powered by a battery. My test here is normal and no problem is found.

pixelwave, 2024-04-30:
@lewisxhe Thanks for the feedback. As I also expected seemed to be a hardware issue. I my case both products I ordered showed the same faulty behavior. Maybe it was a bad batch / slipped through quality control.

Would be nice to receive an updated (V1.2) or quality controlled version. I still like the board / form-factor combination for a lot of use-cases. But the touch needs to work reliably every time when running on battery ...

pixelwave, 2024-06-05:
@lewisxhe 

"The sensitivity of the capacitive touch screen decreases when powered by battery, but works normally when powered by USB. I have reported the problem to the supplier."

Is there a new hardware revision out that solves this issue? I am still very interested in the product.

lewisxhe, 2024-06-05:
You just need to place a new order. All our products are tested before leaving the factory.

ripplebiz, 2024-08-15:
I am experiencing the same touch problems when powered by battery.
My unit was only purchased recently, about beginning of Aug. So, is this faulty touch hardware still an issue?

pixelwave, 2024-08-15:
> I am experiencing the same touch problems when powered by battery. My unit was only purchased recently, about beginning of Aug. So, is this faulty touch hardware still an issue?

What is your hardware revision (if I remember correctly it should be printed on the PCB if you remove the battery)? Mine was "V1.1".

ripplebiz, 2024-08-15:
I have the 'expansion' version, and doesn't have a battery. Can't see the PCB, so not sure what the version is.