Xinyuan-LilyGO/T-Display-S3-Pro #11: Stemma QT Pinout

everybodyfloats, 2024-06-28:
Can you provide the Stemma QT Pinout?

aleonnet, 2024-07-04:
Hi @lewisxhe

Can you please provide an example to use this Stemma QT/UART for connecting an AT6558 navigation chip/GPS?
Would something like this work?

#include <HardwareSerial.h>
static const uint32_t GPSBaud = 9600;
HardwareSerial GPSRaw(2);

void Setup() {
    GPSRaw.begin(GPSBaud);
...
}

void Loop(){
    if (GPSRaw.available()) {
        int ch = GPSRaw.read();
        uiRenderer.drawTextBox(mapSprite, String(ch), 10, 10, 4, TFT_WHITE);
    }
..
}

lewisxhe, 2024-07-04:
Refer to here, specify RX TX before initialization
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/299a71b1f79d5f372edfa2c1e0336e0de47b8b87/examples/QWIIC_GPS_Shield/QWIIC_GPS_Shield.ino#L30

aponert, 2025-05-04:
> Can you provide the Stemma QT Pinout?

Because I stumbled on this. I think the schematic shows it very nicely:

[image]

Also the pinout is printed on the PCB.

So, with the device display side upwards and the stemma qt / qwiic ports pointing towards you, the pinout is:

Left port: U0RXD | U0TXD | 3v3 | GND
Right port: SCL | SDA | 3v3 | GND 

While ESP32-S3 might be flexible assigning pins to alternative functions, I would not recommend to re-assign those pins. I have the strong feeling they are hardwired to the sensors and by reassigning you would loose light sensor, touch and power control.