Xinyuan-LilyGO/T-Display-S3-Pro #16: error compilation

lopsterman, 2024-10-02:
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp: In function 'bool sd_card_bmp_lvgl(const char*, uint32_t, uint32_t, lv_obj_t*)':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:181:57: error: no match for 'operator=' (operand types are 'lv_color_t' and 'lv_color32_t')
  181 |             lv_color = lv_canvas_get_px(canvas, j, h - i);
      |                                                         ^
In file included from c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/draw/lv_draw_buf.h:18,
                 from c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_utils.h:18,
                 from c:/Users/Professional/Documents/Arduino/libraries/lvgl/lvgl.h:35,
                 from c:\Users\Professional\Documents\Arduino\libraries\lvgl\src/lvgl.h:16,
                 from C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.h:9,
                 from C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:3:
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_color.h:101:3: note: candidate: 'constexpr lv_color_t& lv_color_t::operator=(const lv_color_t&)'
  101 | } lv_color_t;
      |   ^~~~~~~~~~
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_color.h:101:3: note:   no known conversion for argument 1 from 'lv_color32_t' to 'const lv_color_t&'
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_color.h:101:3: note: candidate: 'constexpr lv_color_t& lv_color_t::operator=(lv_color_t&&)'
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_color.h:101:3: note:   no known conversion for argument 1 from 'lv_color32_t' to 'lv_color_t&&'
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:182:36: error: 'struct lv_color_t' has no member named 'ch'
  182 |             color.rgb.r = lv_color.ch.red;
      |                                    ^~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:183:37: error: 'struct lv_color_t' has no member named 'ch'
  183 |             color.rgb.g = (lv_color.ch.green_h << 3 | lv_color.ch.green_l) & 0x3f;
      |                                     ^~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:183:64: error: 'struct lv_color_t' has no member named 'ch'
  183 |             color.rgb.g = (lv_color.ch.green_h << 3 | lv_color.ch.green_l) & 0x3f;
      |                                                                ^~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\sd_card.cpp:184:36: error: 'struct lv_color_t' has no member named 'ch'
  184 |             color.rgb.b = lv_color.ch.blue;
      |                                    ^~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino: In function 'void eeproom_init()':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:66:9: error: 'lv_memset_00' was not declared in this scope; did you mean 'lv_memset'?
   66 |         lv_memset_00(eeprom_buf, EEPROM_SIZE_MAX);
      |         ^~~~~~~~~~~~
      |         lv_memset
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino: In function 'bool initSensor()':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:196:9: error: 'class SensorLTR553' has no member named 'setPsLedCurrnet'; did you mean 'setPsLedCurrent'?
  196 |     als.setPsLedCurrnet(SensorLTR553::PS_LED_CUR_100MA);
      |         ^~~~~~~~~~~~~~~
      |         setPsLedCurrent
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino: In function 'void updateLightDected(lv_timer_t*)':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:255:27: error: invalid use of incomplete type 'lv_timer_t' {aka 'struct lv_timer_t'}
  255 |                 if(cnt * t->period > ps_time * 1000){
      |                           ^~
In file included from c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/lv_init.h:17,
                 from c:/Users/Professional/Documents/Arduino/libraries/lvgl/lvgl.h:21,
                 from c:\Users\Professional\Documents\Arduino\libraries\lvgl\src/lvgl.h:16,
                 from C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.h:12,
                 from C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:6:
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_types.h:112:16: note: forward declaration of 'lv_timer_t' {aka 'struct lv_timer_t'}
  112 | typedef struct lv_timer_t lv_timer_t;
      |                ^~~~~~~~~~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:265:27: error: invalid use of incomplete type 'lv_timer_t' {aka 'struct lv_timer_t'}
  265 |                 if(cnt * t->period > ps_time * 1000){
      |                           ^~
c:/Users/Professional/Documents/Arduino/libraries/lvgl/src/misc/lv_types.h:112:16: note: forward declaration of 'lv_timer_t' {aka 'struct lv_timer_t'}
  112 | typedef struct lv_timer_t lv_timer_t;
      |                ^~~~~~~~~~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino: In function 'void camera_init()':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:350:5: error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?
  350 |     ledcSetup(LEDC_WHITE_CH, 1000, 8);
      |     ^~~~~~~~~
      |     ledc_stop
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\factory.ino:351:5: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  351 |     ledcAttachPin(CAMERA_WHITH_LED, LEDC_WHITE_CH);
      |     ^~~~~~~~~~~~~
      |     ledcAttach
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\ui.cpp: In function 'void backgBut_explain(lv_obj_t*, lv_obj_t**, char*, lv_event_cb_t, void*, const void*, lv_coord_t, lv_coord_t)':
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\ui.cpp:331:56: error: invalid conversion from 'int' to 'lv_obj_flag_t' [-fpermissive]
  331 |         lv_obj_add_flag(img_obj, LV_OBJ_FLAG_CLICKABLE | LV_OBJ_FLAG_ADV_HITTEST);
      |                                  ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                        |
[... 495 lines cut ...]
      |     lv_fs_drv_register
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.cpp:149:12: error: 'lv_indev_drv_t' does not name a type; did you mean 'lv_indev_data_t'?
  149 |     static lv_indev_drv_t indev_drv;
      |            ^~~~~~~~~~~~~~
      |            lv_indev_data_t
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.cpp:150:24: error: 'indev_drv' was not declared in this scope
  150 |     lv_indev_drv_init(&indev_drv);
      |                        ^~~~~~~~~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.cpp:150:5: error: 'lv_indev_drv_init' was not declared in this scope; did you mean 'lv_fs_drv_init'?
  150 |     lv_indev_drv_init(&indev_drv);
      |     ^~~~~~~~~~~~~~~~~
      |     lv_fs_drv_init
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.cpp:152:25: error: 'lv_touchpad_read' was not declared in this scope
  152 |     indev_drv.read_cb = lv_touchpad_read;
      |                         ^~~~~~~~~~~~~~~~
C:\Users\Professional\Desktop\T-Display-S3-Pro-master\examples\factory\LV_Helper.cpp:153:5: error: 'lv_indev_drv_register' was not declared in this scope; did you mean 'lv_fs_drv_register'?
  153 |     lv_indev_drv_register(&indev_drv);
      |     ^~~~~~~~~~~~~~~~~~~~~
      |     lv_fs_drv_register
Multiple libraries were found for "SD.h"
  Used: C:\Users\Professional\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\SD
  Not used: C:\Users\Professional\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: no match for 'operator=' (operand types are 'lv_color_t' and 'lv_color32_t')

lopsterman, 2024-10-16:
yes

lopsterman, 2024-10-16:
C:\Users\Professional\Desktop\Новая папка (7)\Si473x_LilyGo_T-Display_Interface_V4_1b-main\Si473x_LilyGo_T-Display_Interface_V4_1b\Si473x_LilyGo_T-Display_Interface_V4_1b.ino:98:10: fatal error: Battery18650Stats.h: No such file or directory
   98 | #include <Battery18650Stats.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: Battery18650Stats.h: No such file or directory

lewisxhe, 2024-10-17:
Si473x_LilyGo_T-Display_Interface_V4_1b.ino ? ?

This repository does not have this example.