Xinyuan-LilyGO/T-Display-S3-Pro #19: unrecognisable issue

sujay119, 2025-01-14:
Hi ,
I’m new to the ESP32 world, and I recently got a LilyGo T-Display S3 Pro board to create a photo viewer that reads images from an SD card.

Initially, I uploaded a simple SD card reading code, and it worked fine. However, after that, the board started showing errors, and now it won’t start at all.

If I try uploading another code, it throws GPIO errors.
If I fix those, it says the T-Display S3 Pro setup is not selected, even though I selected it in the configuration.
I’m stuck and don’t know how to proceed. Any help would be appreciated!

Here’s what I’ve tried so far:

Ensured the correct board is selected in the Arduino IDE.
Verified the User_Setup_Select.h file in the TFT_eSPI library to confirm the correct setup is uncommented.
Reinstalled the drivers and libraries.
My Setup:
Board: LilyGo T-Display S3 Pro
IDE: Arduino IDE 

CODE:
#include <TFT_eSPI.h>
#include <SD.h>

#define SD_CS 5  // Adjust to your SD card CS pin

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);

  // Initialize TFT display
  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);

  // Initialize SD card
  if (!SD.begin(SD_CS)) {
    tft.fillScreen(TFT_RED);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.drawString("SD Init Failed!", 10, 10);
    while (true);
  }

  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.drawString("SD Initialized!", 10, 10);

  // Test: List files on the SD card
  File root = SD.open("/");
  tft.drawString("Files on SD:", 10, 30);
  int y = 50;
  while (true) {
    File file = root.openNextFile();
    if (!file) break;
    tft.drawString(file.name(), 10, y);
    y += 20;
    file.close();
  }
}

void loop() {}
AFTER THIS IT STARTED

sujay119, 2025-01-14:
can you guys please help

sujay119, 2025-01-14:
and if you can provide skype or discord to chat live

lewisxhe, 2025-01-14:
Delete all libraries and re-copy the copy provided by this repository. Do not upgrade the library version.

sujay119, 2025-01-14:
> Delete all libraries and re-copy the copy provided by this repository. Do not upgrade the library version.

ok

sujay119, 2025-01-14:
In file included from c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:97,
                 from c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:16:
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c: In function 'void dc_callback(spi_transaction_t*)':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:203:22: error: 'GPIO' was not declared in this scope
  203 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c:816:28: note: in expansion of macro 'DC_D'
  816 |   if ((bool)spi_tx->user) {DC_D;}
      |                            ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c:817:9: note: in expansion of macro 'DC_C'
  817 |   else {DC_C;}
      |         ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:239:22: error: 'GPIO' was not declared in this scope
  239 |         #define CS_L GPIO.out1_w1tc.val = (1 << (TFT_CS - 32)); GPIO.out1_w1tc.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:80:5: note: in expansion of macro 'CS_L'
   80 |     CS_L;
      |     ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::begin_nin_write()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:239:22: error: 'GPIO' was not declared in this scope
  239 |         #define CS_L GPIO.out1_w1tc.val = (1 << (TFT_CS - 32)); GPIO.out1_w1tc.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:92:5: note: in expansion of macro 'CS_L'
   92 |     CS_L;
      |     ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:240:22: error: 'GPIO' was not declared in this scope
  240 |         #define CS_H GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))//;GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:106:7: note: in expansion of macro 'CS_H'
  106 |       CS_H;
      |       ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::end_nin_write()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:240:22: error: 'GPIO' was not declared in this scope
  240 |         #define CS_H GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))//;GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:121:7: note: in expansion of macro 'CS_H'
  121 |       CS_H;
      |       ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:239:22: error: 'GPIO' was not declared in this scope
  239 |         #define CS_L GPIO.out1_w1tc.val = (1 << (TFT_CS - 32)); GPIO.out1_w1tc.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:141:5: note: in expansion of macro 'CS_L'
  141 |     CS_L;
      |     ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:240:22: error: 'GPIO' was not declared in this scope
  240 |         #define CS_H GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))//;GPIO.out1_w1ts.val = (1 << (TFT_CS - 32))
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:161:7: note: in expansion of macro 'CS_H'
  161 |       CS_H;
      |       ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::writecommand(uint8_t)':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:984:3: note: in expansion of macro 'DC_C'
  984 |   DC_C;
      |   ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:203:22: error: 'GPIO' was not declared in this scope
  203 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
[... cut ...]
 3763 |       DC_C; tft_Write_8(TFT_PASET);
      |       ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\v.sujay\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3769:3: note: in expansion of macro 'DC_C'
 3769 |   DC_C; tft_Write_8(TFT_RAMWR);
      |   ^~~~

exit status 1

Compilation error: exit status 1

sujay119, 2025-01-14:
this gave this error

sujay119, 2025-01-14:
what should i do

lewisxhe, 2025-01-14:
Arduino ESP core must use 2.0.14

lewisxhe, 2025-01-14:
[image]

sujay119, 2025-01-14:
Sketch uses 322865 bytes (10%) of program storage space. Maximum is 3145728 bytes.
Global variables use 19984 bytes (6%) of dynamic memory, leaving 307696 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM7
Connecting....

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

sujay119, 2025-01-14:
again error

lewisxhe, 2025-01-14:
[image]

sujay119, 2025-01-14:
i uploaded the adjust backlight code and the display shows nothing

https://github.com/user-attachments/assets/00643017-fb7d-43e4-a2ef-e0ce123c8787

sujay119, 2025-01-14:
and if i upload another code it says 
Invalid FQBN: getting build properties for board esp32:esp32:esp32s3: invalid option 'ZigbeeMode'

Compilation error: Invalid FQBN: getting build properties for board esp32:esp32:esp32s3: invalid option 'ZigbeeMode'

lewisxhe, 2025-01-14:
What system do you use? If it is Windows, then go to this path, delete the esp32 folder, and then reinstall it C:\Users\YourNmae\AppData\Local\Arduino15\packages

sujay119, 2025-01-14:
thank you so much man really got tensed actually working on project 
can i ask you questions tomorrow  at this time if i account any error

sujay119, 2025-01-16:
i just uploaded the code and the display went blank.didn't show any image.
one small request, i'm a beginner  don't know why the images are not opening can you please guide me how to view images directly from sd card which i stored in it ,when i insert the sd card

sujay119, 2025-01-16:
or can you please mention any platform where we can voice call and discuss ,this will be very help full

lewisxhe, 2025-01-16:
[image]
Do you want to put the pictures into SD?

sujay119, 2025-01-16:
yes like 20-30 pictures

sujay119, 2025-01-16:
i want to store in sd card and when i insert sd card it should show me the images

lewisxhe, 2025-01-16:
The code only accesses the image named 480x222.jpg. Other names will not be accessed.

sujay119, 2025-01-16:
so i should store every image by name of 480x222.jpg

sujay119, 2025-01-16:
then how to access other files

sujay119, 2025-01-16:
is there any way i can see another images

lewisxhe, 2025-01-16:
I only provide a minimal sample application. For the rest, you need to read the SD content yourself to determine the image to access. You need to modify the code yourself.

sujay119, 2025-01-16:
ok ill try out now if i find error ill contact you

sujay119, 2025-01-16:
i read and uploaded the code
#include <TFT_eSPI.h>
#include <FS.h>
#include <SD.h>
#include <Wire.h>
#include <Adafruit_FT6206.h>
#include <TJpg_Decoder.h>

// Define TFT and touch
TFT_eSPI tft = TFT_eSPI();
Adafruit_FT6206 touch = Adafruit_FT6206();

#define SD_CS 10                      // SD card chip select pin
#define SCREEN_WIDTH 222              // Width is 222
#define SCREEN_HEIGHT 480             // Height is 480

String imageDir = "/images";          // Directory where images are stored
String tempImageName = "/480x222.jpg"; // Required image filename
int imageIndex = 0;
String imageList[50];                 // Array to store image file paths
int imageCount = 0;                   // Number of images in the directory

// Callback for drawing JPEG on TFT
bool tftDrawCallback(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t *bitmap) {
  if (y >= SCREEN_HEIGHT) return 0; // Stop drawing if outside display area
  tft.pushImage(x, y, w, h, bitmap);
  return 1;
}

void setup() {
  Serial.begin(115200);

  // Initialize TFT
  tft.init();
  tft.setRotation(3); // Rotate the screen to match the 480x222 orientation

  // Initialize touch
  if (!touch.begin()) {
    Serial.println("Touch initialization failed!");
    while (1);
  }

  // Initialize SD card
  if (!SD.begin(SD_CS)) {
    Serial.println("SD card initialization failed!");
    while (1);
  }

  // Initialize JPEG decoder
  TJpgDec.setJpgScale(1); // No scaling
  TJpgDec.setCallback(tftDrawCallback);

  // Load list of images
  loadImages();

  // Display the first image
  if (imageCount > 0) {
    displayImage(imageList[imageIndex]);
  } else {
    Serial.println("No images found in the directory!");
  }
}

void loadImages() {
  imageCount = 0;
  File root = SD.open(imageDir);

  if (!root) {
    Serial.println("Failed to open image directory!");
    return;
  }

  File file = root.openNextFile();
  while (file && imageCount < 50) {
    if (!file.isDirectory()) {
      String fileName = file.name();
      if (fileName.endsWith(".jpg")) {
        // Replace spaces and special characters in the file name
        fileName.replace(" ", "_");  // Optional: Replace spaces with underscores
        fileName.replace("(", "");   // Remove any parentheses
        fileName.replace(")", "");   // Remove any parentheses
        imageList[imageCount++] = fileName;
        Serial.println("Found image: " + fileName);  // Log the found image name
      }
    }
    file = root.openNextFile();
  }

  root.close();
  Serial.printf("Found %d images in the directory.\n", imageCount);
}

void displayImage(const String &path) {
  // Rename the selected image to the required filename
  if (SD.exists(tempImageName)) {
    SD.remove(tempImageName);
  }

  if (SD.rename(path, tempImageName)) {
    // Display the image using the renamed file
    tft.fillScreen(TFT_BLACK);
    if (TJpgDec.drawJpg(0, 0, tempImageName) != 0) {
      Serial.println("Error decoding JPEG image!");
    }
  } else {
    Serial.println("Failed to rename image to temporary name!");
  }
}

void handleTouch() {
  if (!touch.touched()) return;

  TS_Point p = touch.getPoint();
  int x = p.x;

  if (x < 50) {
    // Swipe left
    if (imageIndex > 0) {
      imageIndex--;
      displayImage(imageList[imageIndex]);
    }
  } else if (x > SCREEN_WIDTH - 50) {
    // Swipe right
    if (imageIndex < imageCount - 1) {
      imageIndex++;
      displayImage(imageList[imageIndex]);
    }
  }
}

void loop() {
  handleTouch();
}

sujay119, 2025-01-16:
it gave me this image after uploding

sujay119, 2025-01-16:
[image]

sujay119, 2025-01-17:
ONE SMALL DOUBT I REPLACED THE BATTERY  WITH A 1200mah battery the battery health indicator is fluctuating once shows 100, once 32,once 0(still working) . is it the issue of the battery

lewisxhe, 2025-01-20:
Check the battery connector for contact problems

sujay119, 2025-06-25:
@lewisxhe the display of the lilygo s3 pro got stuck what can I do to bring it functionback i was using it as usual but suddenly the screen got stuck .It's getting turned on but touch display is not working, though I handled it very carefully

lewisxhe, 2025-06-27:
You can try to scan the I2C bus to see if you can see the touch device