Xinyuan-LilyGO/T-Display-S3-Pro #20: Power switch stops working to turn on s3 pro

HoaxParagon, 2025-02-09:
I've enabled the pmu and made it enable charging but the problem is that without a usb-c connector to start the thing it doesn't work. The power switch doesn't do anything any more. I've uploaded custom code to the thing, more specifically the ESP32 Marauder software that I ported to the S3 pro. I'm not done porting it yet but now the button doesn't work to power the unit.

HoaxParagon, 2025-02-13:
I think it's something in software because if the switch is on when the usb-c is unplugged, it stays on. Must be something in software.

lewisxhe, 2025-02-13:
If so, then you have to consider this, are you using ArduinoIDE?? If yes, then set USB CDC ON BOOT to Disable

[image]

HoaxParagon, 2025-02-13:
I've done this and reuploaded via arduino IDE, and the button still won't turn on the unit.

lewisxhe, 2025-02-13:
Then there is only one way to test it. You need a USB2TTL, connect it to the Qwiic 43 (TX), 44 (RX) of DISPLAY, and monitor whether the ESP is started.

Remember the connection of RX and TX, to be opposite,
USB2TTL RX -> DISPLAY TX
USB2TTL TX -> DISPLAY RX