Xinyuan-LilyGO/T-Display-S3-Pro #22: LilyGo S3 Pro Camera - does connect / disconnect to laptop (every 3 sec) - does not boot

Kingair001, 2025-04-25:
Dear,
I bought a LilyGo S3 Pro Camera via AliExpress and the people asked me to contact you.
My LilyGo S3 Pro Camera does not boot, it does connect / disconnects every 3 seconds from my laptop
How can I solve this please ?

lewisxhe, 2025-04-25:
You should check if your Arduino IDE options match the example. Constant disconnection is a sign of program crash.
https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro#6%EF%B8%8F%E2%83%A3-faq

Kingair001, 2025-04-25:
I have a black screen, nothing is happening
I am not Bill Gates/ into IT, so please explain each step
I bought it new
Plugged it in, and this is what happens

lewisxhe, 2025-04-25:
You just got it?

Haven't uploaded any sketches yet?Just got it?

Haven't uploaded any sketches?

You can try uploading any factory firmware sketch to see if the screen can display normally
https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/firmware/v1.1

Kingair001, 2025-04-25:
Not just but I was waiting on the people frol LilyGo what to do
They adviced me to open an Issue here

Now it connects/disconnects every 3 seconds, so how will I be qable to upload anything?

lewisxhe, 2025-04-25:
You need enter esp to download mode

1. Connect the board via the USB cable
2. Press and hold the BOOT button , While still pressing the BOOT button, press RST
3. Release the RST
4. Release the BOOT button
5. Upload firmware

Follow the above sequence, and the USB port will be fixed and no longer flash.

Download this [firmware](https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/blob/master/firmware/v1.1/UnitTest_V1.1_DisChg_Debug.bin)

Follow this README and upload the firmware to the esp device through esp download tools
https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/tree/master/firmware

After uploading, press the RST button to restart the device. The screen should show

Kingair001, 2025-04-25:
I can not download the flash tool -> error 403 foridden !
I have the firmware you provided via the link

lewisxhe, 2025-04-25:
OK, I can download it. Or you can use the online download tool, check here, it is the same method, please note that the write address must be 0
https://espressif.github.io/esptool-js/

Kingair001, 2025-04-25:
Nothing happens, it does still the same: connect / disconnect 
Do you have a link for the flash firmware tool, so I can try that ?

lewisxhe, 2025-04-25:
First step If I put the ESP into download mode, does it still disconnect and connect repeatedly?

Kingair001, 2025-04-25:
Yes it does !

lewisxhe, 2025-04-25:
Still? ?

Then it may be that the steps are incorrect.
Please confirm whether you have followed the steps below

[image]

Kingair001, 2025-04-25:
Now red light flickers - really fast

lewisxhe, 2025-04-25:
The red light is just a battery indicator, it will flash when there is no battery. Have you entered the download mode correctly now?

Kingair001, 2025-04-25:
It keeps doing the same 
Connect / disconnect 
I hear the sound every 3 sec

lewisxhe, 2025-04-25:
OK, please record a video of the device failing to enter download mode, send it to the dealer to confirm that it is damaged, and then they will handle the after-sales replacement package for you.

Kingair001, 2025-04-25:
I already did that, I'll send them a screenshot of this conversation 
Hopefully it get's solved 
Thanks

lewisxhe, 2025-04-25:
Yes. The dealer will communicate with you later.